micro-ROS/NuttX #1: Problem to run I2C on Olimex-STM32-E407

jfm92, 2018-02-06:
Hi,
I'm triying to run the BMP180 sensor (The reason to choose this one, is because the driver is already done and checked that it works)with the Olimex Board just to figure out how to use properly I2C on Nuttx but at the moment I wasn't be able to make it run.

To make it run I've done the next configurations and modifications:
**-I set this configuration profile olimex-stm32-e407/nsh.**
**-Then inside the menuconfig I checked the next options:**
*System Type -> STM32 Peripheral Support -> I2C1 (Checked)
*Device Drivers -> I2C Driver Support (Checked)
*Device Drivers -> Sensor Device Support (Checked) -> Bosch BMP180 (Checked)
*Application Configuration-> Examples-> BMP180 Sensor Example(Checked)(This last one it's optional, but make easier the way to use the sensor in the console)

Once we have set the configuration, I copied the file stm32_bmp180.c from nuttx/configs/stm32f103-minimun/src to nuttx/configs/olimex-stm32-e407/src and I modify `#include "stm32f103_minimum.h"` to `#include "olimex-stm32-e407.h"`

At this point, every file is in nuttx/configs/olimex-stm32-e407/src

**In Makefile add this line:**
[image]

**In olimex-stm32-e407.h add this line:**
[image]

Now it should compile without any problem.
I asked in the Nuttx Forum about if the configuration is correct and if I've maded the right modification and they tell me that is ok.
https://groups.yahoo.com/neo/groups/nuttx/conversations/topics/17116

So when you run the BMP180 example in the console, you don't get any data so I decided to debug with GDB

jfm92, 2018-02-09:
This problem is already solve.
I will write a detailed guide in other issue explaining how to make it run .
Close.

closed 2018-02-09 by jfm92

Viplav04, 2019-10-02:
Hello Juan @jfm92 
I am trying this example, but my device is not registered. I do not see /dev/press0 when I execute ls/dev in NSH console. I want to wrap this sensor in a micro-ROS node. You wrote you solved the problem.
Can you share the solution?