micro-ROS/NuttX #12: How to create your custom App (Under Construction)

jfm92, 2018-02-23:
When you develop an App for NuttX, you can choose if your App is an example, a system utility, can utility..
The only difference beteween the diferents options, is in which folder you save your app. 
To this guide, we're going to develop the App in the example folder.

So go to: `App/Examples` 
And create a folder with the name of your App (In this case it'll be custom_app).
Each App have four basicall file:
[image]

- **custom_app_main.c** : This is the main file of the project. This file have acces to:
  - Header file of the driver.
  - Basic C/C++ library.
  - Basic NuttX library.
  - You can access to other files that you add to this folder.
- **Kconfig** : Here you programm the options that will appear at the menuconfig, such as the description, the name, dependencies...
- **Make.defs** : This file add this application when it's check in menu.
- **Makefile** : This file is use to add the exactly file to the compilation and which options do you want to have in the App and in the system when you're run it.

### Kconfig
The next image is the configuration of this App in the menu:
[image]

You can add some configuration options to your app that it'll appear in the menuconfig:
[image]
In this case you can see the options of the priority of the task when it's running the App and the stack size.

### Make.defs

closed 2018-02-23 by jfm92

darauble, 2019-04-17:
Hi, I've run into some other problem, for which I cannot find a description, but this post seems relevant.

Let's say, I have two custom apps: APP1 and APP2. APP2 has underlying functionality of APP1, but different frontend. So in APP2 I'd like to include "app1.h", i.e. reuse source of APP1. Currently, of course, compilation fails, as "app1.h" is inaccessible for APP2.

How could I reuse sources from one app in another? Let's say, I don't care about actually linking same object files twice (no need for shared library). How can this be achieved? How to make APP1's headers/sources visible "publicly", what is the _correct_ way of doing this? (The way of putting reused source elsewere is _not an option_, only in APP1's directory.)

imuguruza, 2019-04-17:
@jfm92 may be you could help here
I suggest you also checking NuttX [page](https://nuttx.org/doku.php?id=documentation) and its [google group](https://groups.google.com/forum/#!forum/nuttx).

darauble, 2019-04-17:
I've come up with a solution like this:

CFLAGS += -I $(APPDIR)/examples/APP1

However, this does not look very nice for me, even though... it works.