micro-ROS/NuttX #20: How to use CAN Bus (WIP)

jfm92, 2018-03-14:
There are 2 possibilities to use CAN in NuttX:
Using and adapter like the MPC2515 which works thought SPI.
If your board have native CAN you can use directly with your board.
Once is configured the peripheral, for the APP layer it doesn't matter which are you using, because the behavior is the same.
## Configuration of an external module
In this case we're going a module how integrates the popular MPC2515 Chip.
-First it's neccesary to do the wiring properly connecting. If your board have support for this chip, you can see the pin in the next files:
**MOSI,MISO,CLK**:  NuttX/configs/<your_board>/include/board.h
**CS,Interruption**: NuttX/configs/<your_board>/src/<your_board>.h
(The CS it's usually in the board.h file, but in this case it's a little different)

Once the wiring it's ready, in the menuconfig, we must set the next options:
_System Type-> STM32 Peripheral Support ->SPI1_
_Device Drivers->CAN Driver Support->Microchip MCP2515 CAN Bus Controller over SPI_
_Application Configuration-> Can Utilities->CAN utility library_

## Native CAN Bus.
(Work in progress)

## APP
Application Configuration-> CAN example 
With this APP, you must choose between read or wirte. The option Read-Write it's doens't work properly. 

**It's neccesary to say that this APP doesn't work properly, so it's usefull to understand how to work with this peripheral, but not to use it.**

Also here you can see a little summary of how to use the driver in you APP. 
(Work in progress)

bhavesh-sr, 2021-07-26:
@jfm92 Did you have any luck making the Native CAN Bus work? I have Tx working but not Rx, and I'm not sure why. For Tx I can see messages on the line and I have a separate receiving computer which can read those messages. But for Rx, I am unable to receive anything using the CAN app example. Is it the app that's broken or the driver?

P.S. I am running on an Olimex STM32-E407 board

dayjaby, 2021-08-30:
@bhavesh-sr I am using micro-ROS with Zephyr OS for a CAN application. Did you consider that OS yet?

bhavesh-sr, 2021-08-30:
@dayjaby We also ended up switching our OS. We are using FreeRTOS now and CAN works well for both Tx and Rx.