micro-ROS/NuttX #21: Serial on Sim configuration

BorjaOuterelo, 2018-03-27:
Hi

I am trying to generate an example of UART serial communication using sim configuration but I am not being able to link nuttx binary.
I got the following error:
```
LD:  nuttx
nuttx.rel: In function `open_serial':
(.text+0x4a094): undefined reference to `NXtcgetattr'
nuttx.rel: In function `open_serial':
(.text+0x4a0da): undefined reference to `NXtcsetattr'
collect2: error: ld returned 1 exit status
```
Does sim configuration support UART?. How do I activate UART for sim configuration? I have gone through Kconfig menu several times but haven't found the right option there.

Thanks.

vmayoral, 2018-03-27:
Hello @BorjaOuterelo,

We need to look at this more closely but quick feedback: I don't think you'll be able to use UART from the simulator. Instead, I'd recommend trying the [reference hardware platform](https://github.com/microROS/hardware) (OLIMEX board).

jfm92, 2018-03-27:
HI @BorjaOuterelo ,
The NuttX simulator doesn't have access to the PC peripheral.
I think it's useful to check internal SO features like timers, threads...

BorjaOuterelo, 2018-03-27:
Hi @vmayoral  and @jfm92 
So the simulated UART device mentioned in http://nuttx.org/doku.php?id=wiki:nxinternal:simulator is not usable as a UART from within my NuttX app?

jfm92, 2018-03-27:
Well in the document suggest someway to do a like a simulation of an UART connection, in the "Toward a design" point. 
Anyways I think it's not implemented yet and also I don't recommend to work with the simulator, because if you look a the [readme file](https://bitbucket.org/nuttx/nuttx/src/d393f43caa3f280e33fe303840fc19aac07200cf/configs/sim/README.txt?at=master&fileviewer=file-view-default) of the simulator there are so few things working properly there.

closed 2018-06-11 by jfm92