micro-ROS/NuttX #22: Toolchain to be used

BorjaOuterelo, 2018-03-27:
Hi.

I am building my application using stm32-e407 provided docker file.
On that file I found:
```
RUN git clone https://bitbucket.org/nuttx/buildroot
# The following error is obtained when using this version:
    # cfns.gperf:101:1: error: ‘gnu_inline’ attribute present on ‘libc_name_p’
    # cfns.gperf:26:14: error: but not here
# RUN cd buildroot && cp configs/cortexm3-eabi-defconfig-4.6.3 .config
RUN cd buildroot && cp configs/cortexm3-eabi-defconfig-4.8.5 .config

# RUN ls
# clone the code
RUN git clone https://github.com/microROS/apps
RUN git clone https://github.com/microROS/NuttX nuttx

# select the configuration of the OLIMEX board
RUN cd nuttx && tools/configure.sh olimex-stm32-e407/nsh

#######################
# Build toolchain
#######################
# copy a script to automate the command "make oldconfig" (interative)
# the command has been generated by the instruction:
#             autoexpect make oldconfig
COPY script.exp /root
RUN chmod +x /root/script.exp
# instead of doing this, we'll use a pre-generated script
# RUN cd buildroot && make oldconfig
RUN cd buildroot && /root/script.exp

# get dependencies required to build the toolchain
RUN apt-get install gettext -y wget
RUN apt-get update && apt-get install -y build-essential bison flex gperf libgmp-dev libmpc-dev libmpfr-dev libisl-dev binutils-dev libelf-dev libexpat-dev
# Build the toolchain, fix with the paths, seems specific to this board
RUN mv buildroot nuttx/buildroot
RUN cd nuttx/buildroot && make
```

These lines are building the buildroot for the target.
A couple of lines underneath:
`RUN export PATH="/ ":$PATH && cd nuttx && make`
is found. This builds nuttx and I think the intention is building It with the freshly built buildroot but if I change `make` for `make V=2` I found the following output:
```
make[2]: Entering directory '/root/apps/nshlib'
CC:  nsh_init.c
arm-none-eabi-gcc -c -fno-builtin -Wall -Wstrict-prototypes -Wshadow -Wundef -Os -fno-strict-aliasing -fno-strength-reduce -fomit-frame-pointer -mcpu=cortex-m4 -mthumb -mfloat-abi=soft -I. -isystem /root/nuttx/include   -pipe -I "/root/apps/include"  nsh_init.c -o  nsh_init.o
```
On that output something looks wrong for my undestanding: `arm-none-eabi-gcc` that is not any of the tools I have on the intended toolchain root "/root/nuttx/buildroot/build_arm_nofpu/staging_dir/bin/"
Is this correct? Which toolchain is the one that I shall be using to compile and link an external library? arm-none-eabi or arm-nuttx-eabi?

Cheers

imuguruza, 2018-03-27:
Hi @BorjaOuterelo you should use `arm-none-eabi-gcc` cross-compiler to build. Is the standard toolchain for STM ARM Cortex devices

imuguruza, 2018-03-27:
update comment

closed 2018-06-11 by jfm92