micro-ROS/NuttX #50: Problems in trying to use nuttx and micro-ROS for  EtherCAT device control

KevinHwangAi, 2020-04-14:
Hello everyone.
I am trying to make my robot control module with stm32f767 + nuttx + mirco_ROS + soem.
So far,I have migrated the soem library to nuttx and compiled it successfully.

The project can detect the slave station of EtherCAT and successfully exchange data with it.
However, when PDO data is exchanged with the slave station, the data response from the slave station cannot be received,
After using Wireshark tool,it is found that the slave station has received PDO data and returned WKC = 1,but stm32f767 has not received data.

After the comparison test with Linux system and found that:
On the Linux system, use the test program in the soem library to complete the ec_send_processdata()(the function is used to process the data format and sent to the slave station). 

The time is:
start_time : 192876 s : 335065988 ns
end_time : 192876 s : 335323327 ns

Using the same test program on nuttx system, the time to complete the ec_send_processdata() is:
start_time : 15 s : 150000000 ns
end_time  : 15 s : 160000000 ns

The test result shows that, it takes a long time to complete ec_send_processdata() in nuttx. 
So the stm32f767 missed the time to receive the response from the slave station.

I would like to raise a few questions about nuttx:

1. Does nuttx send and receive raw socket have buffer mechanism for the receive() function get  response data in Ethernet DMA?

2. As far as I know, Stm32f767  can port soem Library in Hal and  communicate with EtherCAT slave station as master station.
However, after using nuttx, stm32f767 can't complete the communication with slave station. 
Is that a lot operation resource of Stm32f767 used by running nuttx system?

3. Is there any way for stm32f767 + nuttx + mirco ﹣ ROS + soem to be implemented? For example, can we shorten the operation time of ec_send_processdata() function by setting nuttx parameters in menuconfig? Add the raw socket buffer mechanism? or other way?

look forward to your help.It would be very cool thing if we could make mirco_ROS and nuttx to control the EtherCAT equipments.

iluetkeb, 2020-04-15:
How did you measure the time? Please be aware that clock_gettime in NuttX doesn't have a very high resolution. It depends on the MCU exactly how much, but the elapsed time above is exactly 10ms, which is kinda suspicious and could indicate that the RTC resolution is the issue.

In general, benchmarking computation times on MCUs is better done by counting cycles. This is unfortunately not trivial on NuttX, but maybe @tkolcon and his team have a good suggestion.

iluetkeb, 2020-04-15:
Regarding your other questions, yes NuttX can buffer network communication. I don't know the exact name of the setting, but try digging around in menuconfig a bit to find it.

KevinHwangAi, 2020-05-01:
After days of testing, no way has yet been found to control the device because the nuttx system does not support ethercat.

closed 2020-05-01 by KevinHwangAi