micro-ROS/NuttX #51: How are changes from nuttx mainline merged into this fork?

cmorganBE, 2020-04-22:
Hello! I'm looking to use micro-ROS on a mcu with our ros2 system. I've run into a couple of issues that thankfully only took a few minutes to identify the cause of, that turned out to have been fixed in early 2019 on nuttx mainline. What approaches are used to keep in sync with nuttx mainline? I'll gladly open PRs and pull over the commits that resolve, I'm more interested in the overall approach to keeping things in sync and avoiding time spent fixing issues already fixed in mainline.

Regards,
Chris

jfm92, 2020-04-22:
Hi Chris,

First of all, thanks for being interested in micro-ROS!
Regarding synchronization with the NuttX mainline, we schedule a rebase with the mainline repo every 6 months. But this schedule was a little delayed because now we support three different RTOS.
So, we're planning to perform a rebase to version 8.2 on the next months, but if you want to do a PR, we will highly appreciate!

Regards,
Juan

cmorganBE, 2020-04-22:
Hi @jfm92. Thank you for the response. Let me look at a rebase against nuttx mainline as an initial step. Imo that would be a more complete effort than cherry picking a few fixes while potentially leaving a lot more behind.

Would you mind if I opened a PR against the website to document the rebase approach and normal schedule?

What kinds of tests would/should I run when doing that rebase before opening up a pr?

FranFin, 2020-07-20:
@cmorganBE Hi! Thank you for your interest in micro-ROS. Notice that we’ll discuss the recent contributions and developments in the next [Embedded WG meeting](https://discourse.ros.org/t/ros-2-embedded-wg-meetings/15460). You’re invited to bring your micro-ROS related issue to this platform. We’ll be very happy to have you join us and discuss your questions with the community.