micro-ROS/NuttX #53: Getting undefined reference when compiling for board stm32f4discovery

Aetherbase, 2020-08-09:
### Steps to reproduce:
`
ros2 run micro_ros_setup create_firmware_ws.sh nuttx stm32f4discovery
ros2 run micro_ros_setup configure_firmware.sh uros
`
other configuration steps as given [here](https://micro-ros.github.io//docs/tutorials/core/first_application_rtos/nuttx/)
Followed by :
`
ros2 run micro_ros_setup build_firmware.sh
`
which produces error given as follows:
`
arm-none-eabi-ld: /home/uros/microros_ws/firmware/NuttX/staging/libarch.a(up_initialize.o): in function `up_initialize':
up_initialize.c:(.text+0x26): undefined reference to `up_netinitialize'
arm-none-eabi-ld: /home/uros/microros_ws/firmware/NuttX/staging/libapps.a(librcutils_logging.c.obj): in function `rcutils_logging_console_output_handler':
/home/uros/microros_ws/firmware/mcu_ws/uros/rcutils/src/logging.c:917: undefined reference to `isatty'
arm-none-eabi-ld: /home/uros/microros_ws/firmware/NuttX/staging/libapps.a(librmw_microxrcedds_rmw_init.c.obj): in function `rmw_init':
/home/uros/microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_init.c:197: undefined reference to `tcgetattr'
arm-none-eabi-ld: /home/uros/microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_init.c:232: undefined reference to `cfsetspeed'
arm-none-eabi-ld: /home/uros/microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_init.c:233: undefined reference to `cfsetspeed'
arm-none-eabi-ld: /home/uros/microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_init.c:235: undefined reference to `tcsetattr'
make[1]: *** [Makefile:185: nuttx] Error 1
make[1]: Leaving directory '/home/uros/microros_ws/firmware/NuttX/arch/arm/src'
make: *** [tools/Makefile.unix:422: pass2] Error 2
`
# System Configuration used:
Ubuntu Focal 20.04
ROS2 Foxy
micro-ROS-build commit id 8c5eb51
****
please help me on getting past this error.

iluetkeb, 2020-08-10:
Can you try using the `drive_base` configuration?

Aetherbase, 2020-08-10:
@iluetkeb  I did it on my first trial, recieved the following:
```
Configuring firmware for nuttx platform stm32f4discovery
Configuration configs/stm32f4discovery/drive_base (expanded from drive_base) not found
```
upon searching, found a github thread where you suggested to use `uros` configuration.

Aetherbase, 2020-08-12:
@iluetkeb How should I proceed?

Aetherbase, 2020-08-17:
@iluetkeb im sorry if im disturbing you again, but if it would be really helpful if you could give me a few guidelines on how to port `drive_base` configuration for stm32f4discovery or how to fix the `uros` configuration. Thank you in advance.

iluetkeb, 2020-08-19:
I'm on vacation, so I have limited Internet access. What you generally need to is run the nuttx menuconfig (source dev_ws/install/setup.bash and then go into "nuttx" and run "make menuconfig), and add the options for termios.

You can also copy over the relevant lines from the drive_base/defconfig file, and be sure to reconfigure afterwards, but that's more the expert way.

Aetherbase, 2020-09-03:
@iluetkeb I can confirm reproducing what @thihakyawjob has mentioned, enabling termios does not fix the undefined reference issue.

FranFin, 2020-09-04:
Hi @Aetherbase. I'm afraid we do not have official support for NuttX on the STM32F4discovery yet, and it's not among our first priorities at the moment. However, we'd be glad to help with concrete technical problems or issues that you may find while trying to give support for it, if this is the plan. PRs are welcome!

Aetherbase, 2020-09-05:
Hello @FranFin I'll be glad to contribute for porting `drive_base` configuration to stm32f4discovery. I just need a little guidance on porting and will do a PR for it.

arunlee77, 2020-10-19:
I am trying the same on nucleo-h743zi and am facing the same issue. Do you have any suggestions to try out?.

I have termios enabled and i am able to build everything successfully, only IF i DISABLE micro-ROS Ping-Pong in the Application Configuration ---> Examples.

arm-none-eabi-ld --entry=__start -nostartfiles -nodefaultlibs -g -T/root/uros_ws/firmware/NuttX/configs/nucleo-h743zi/scripts/flash.ld -L"/root/uros_ws/firmware/NuttX/staging" -L"/root/uros_ws/firmware/NuttX/arch/arm/src/board"  \
arm-none-eabi-ld: /root/uros_ws/firmware/NuttX/staging/libapps.a(nsh_fscmds.o): in function `cmd_mkfifo':
/root/uros_ws/firmware/apps/nshlib/nsh_fscmds.c:1332: undefined reference to `mkfifo'
arm-none-eabi-ld: /root/uros_ws/firmware/NuttX/staging/libapps.a(librcutils_logging.c.obj): in function `rcutils_logging_console_output_handler':
/root/uros_ws/firmware/mcu_ws/uros/rcutils/src/logging.c:917: undefined reference to `isatty'
arm-none-eabi-ld: /root/uros_ws/firmware/NuttX/staging/libapps.a(librmw_microxrcedds_rmw_init.c.obj): in function `rmw_init':
/root/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_init.c:197: undefined reference to `tcgetattr'
arm-none-eabi-ld: /root/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_init.c:232: undefined reference to `cfsetspeed'
arm-none-eabi-ld: /root/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_init.c:233: undefined reference to `cfsetspeed'
arm-none-eabi-ld: /root/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_init.c:235: undefined reference to `tcsetattr'
arm-none-eabi-ld: /root/uros_ws/firmware/NuttX/staging/libapps.a(libmicroxrcedds_client_serial_transport_posix.c.obj): in function `uxr_read_serial_data_platform':
/root/uros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/profile/transport/serial/serial_transport_posix.c:45: undefined reference to `poll'
make[1]: *** [Makefile:185: nuttx] Error 1
make: *** [tools/Makefile.unix:422: pass2] Error 2